AutoFlowResearch/SmartPeak #292: Build fails on Ubuntu 20.04 with latest version

poshul, 2020-12-01:
Building SmartPeak2 on Ubuntu fails as of ba21b7d395be591c798b1973d10c8a7872f465ae. At least part of this seems to be that https://github.com/dmccloskey/SmartPeak2/blob/ba21b7d395be591c798b1973d10c8a7872f465ae/CMakeLists.txt#L20 is commented out so imgui doesn't get downloaded if it doesn't already exist.

ahmedskhalil, 2020-12-02:
This version of SmartPeak2 requires the ImGui version supplied in the `superbuild` folder to be used, what's the exact error you get?

junzwuu, 2020-12-09:
Hi, I found an additional '#' in Smartpeak/CMakeLists.txt line 20 which lead to the failure of imgui installation. However, another error occurred during cmake on smartpeak_build: 
> CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
>   Could NOT find ImGui (missing: Examples)
> Call Stack (most recent call first):
>   /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
>   cmake/modules/FindImGui.cmake:233 (find_package_handle_standard_args)
>   src/tests/class_tests/smartpeak/CMakeLists.txt:97 (find_package)

Seems cmake can locate imgui but can not find components in imgui. There is _examples_ folder in _$HOME/SmartPeak_superbuild/Dependencies/Source/imgui_. 

Thank you.

ahmedskhalil, 2020-12-12:
> Hi, I found an additional '#' in Smartpeak/CMakeLists.txt line 20 which lead to the failure of imgui installation. However, another error occurred during cmake on smartpeak_build:
>
>
> Seems cmake can [...]

For this exact commit please don't overwrite or create your own superbuild directory, the ImGui version supplied inside that folder is the correct one to use.

closed 2021-01-11 by ahmedskhalil