AutoFlowResearch/SmartPeak #347: Isotopic corrections not applied

Staifen, 2021-03-31:
**Describe the bug**

When looking closer to the MDVs I get from the workflow, I realized that the final output of the workflow is not corrected for the derivatization agent (through the isotopicCorrection function). I made a test by running in SmartPeaks a small subset of data with the GCMS fullscan worfkflow and then CALCULATE_MDVS or CALCULATE_MDVS+ISOTOPIC_CORRECTIONS (see attached exported features as pivot tables). As you can see the exported peak_apex_intensity is identical. 

**To Reproduce**
Steps to reproduce the behavior:
see description and attached files

**Expected behavior**
peak_intensity_apex should be changed (and saved somewhere) after the ISOTOPIC_CORRECTIONS function is applied

**logs**
Attach log file. Log, by default, can be found in:
 - Windows: C:\Users\user_name\AppData\Local\SmartPeak
 - Linux: /home/user_name/.SmartPeak
 - MacOS: /Users/user_name/.SmartPeak

**Screenshots**
If applicable, add screenshots to help explain your problem.

**data**
O:\AutoFlow\AutoFlow data\Test_SmartPeak\AA_test_isotopic_correction.zip

**Version information**
 - Windows
 - Cutsom version by Ahmed

ahmedskhalil, 2021-06-17:
Thanks for the submission, could you please provide the tracer used for each of the `ProteinName` in the [traML](https://github.com/AutoFlowResearch/SmartPeak/blob/0f6c6440f96508a8b30fbd950831d395deb1344f/src/examples/data/ProAAs_GCMS_fullscan/traML.csv) file ?

ahmedskhalil, 2021-07-01:
The isotopic tracers used in the experiments such as 13C, 15N.

Staifen, 2021-07-02:
For the 13C experiments we did some time ago, the isotopic tracers were 80/20 ratio of 1-13C/U-13C glucose; [1,6-13C]glucose; [1,2-13C]glucose. However, tracers do not impact how the isotopomer distribution correction is applied; moreover, the different tracers that were used might change in future experiments. I would leave this information at the discretion of the user.

ahmedskhalil, 2021-07-03:
Without the exact isotopic tracer for each ProteinName the correct matrix cannot be generated.

Staifen, 2021-07-23:
I would then suggest to use dummy values for the tracers. The specific isotopic tracers used in the various experiments do not have an impact on how the measured molecules should be corrected. Introducing the specific tracers used in each different sample would complicate a lot the data entry. We can have a brief chat if this is not possible.