AutoFlowResearch/SmartPeak #361: Can not load chromatogram plot of processed data

Staifen, 2021-04-28:
**Describe the bug**
Cannot load Chromatogram plot of processed data. According to Douglas this would should be working using the following workflow: "LOAD_RAW_DATA" "LOAD_FEAUTURES" "MAP_CHROMATOGRAMS". 

**To Reproduce**
Steps to reproduce the behavior:
1. File --> Load session from sequence --> load any sequence.csv of mzML files that were already processed with a workflow in SmartPeak (in example featureXML files created with the "HPLC UV unknowns" or GCMS Full Scan Unknowns").  
2. Create a new workflow: Add Step --> "Raw data methods" --> "LOAD_RAW_DATA" "LOAD_FEAUTURES" "MAP_CHROMATOGRAMS"
3. View --> Chromatogram --> tick any sample/transition to display the relative chromatogram..

**Expected behavior**
Be able to see chromatograms of integrated data

**logs**
Log file attached, related to the data linked.

**data**
You can use data from here [O:\AutoFlow\AutoFlow data\Test_SmartPeak\test_bug] but I guess should apply to any kind of processed data..

**Version information**
 - OS: [Windows 10]
 - Version [e.g. 1.3.0]

bertrandboudaud, 2021-04-28:
Hello Staifen

Thanks for the complete report;

There are some limitation with the chromatogram that if too much points are to be displayed, it will not appear. normally it is reported with some error message that i don't see there. however, if we reduce the scope of the chromatogram we can see the chromatogram as shown is this gif:

[image]

as we can see there while increasing the window scale, the chromatogram disapear, which seems to confirm that we are reaching the limitation of number of points to draw.

note that this situation will be more explicit with the next release (see pull request https://github.com/AutoFlowResearch/SmartPeak/pull/359)

This is a first feedback; i will check the behavior with the next coming version;

Staifen, 2021-04-28:
Hi Bertrand,

Thank you for your reply. Question: did you manage to display those
chromatograms just by loading the feature files as I mentioned (and without
re-running a peak integration workflow)? Because it does not work on my PC.

Best,
Stefano

bertrandboudaud, 2021-04-28:
I used the workflow that you have described here "LOAD_RAW_DATA" "LOAD_FEAUTURES" "MAP_CHROMATOGRAMS"; then I also run this workflow (which you don't mention there, but it is a necessary steps to load the data right?).

maybe another difference, I run all this from local hard drive.

Also, the capture has been made with Smartpeak version 1.2.0 but I also tried with SmartPeak 1.3.0 and it work as well.

Let me know if you see a difference between your execution and mine I can double check.

Staifen, 2021-04-28:
Do you mean the "HPLC unknown" workflow? Running that workflow will
reprocess the whole dataset, by loading the raw data and integrating the
peaks. In principle that would be fine, but running the whole workflow
takes a very long time (without using CLEAR_DATA at the end of the
workflow), especially when running it with a long sample list.

Maybe to be more clear, what I need to do is to validate a peak picking
method, and I want to verify it on many samples >50 to be sure it is
robust. To run the "HPLC unknown" workflow on that, my PC takes hours. Then
I would need to check manually if everything is integrated
correctly (looking at the chromatogram plots), and in case change some
parameters and run the whole "HPLC unknown" workflow again. Douglas
suggested that I use the CLEAR_DATA function at the end of "HPLC unknown"
to make sure the RAM is not overloaded, and then load the feature files.
That makes the workflow fast, but then I do not obtain the chromatogram
plots. And loading the features of the processed raw data files does not
lead to chromatogram plots.

I am running SmartPeak on Windows.. could that be the reason?

bertrandboudaud, 2021-04-28:
No, I mean I create the workflow as you describe, with the 3 steps "LOAD_RAW_DATA" "LOAD_FEAUTURES" "MAP_CHROMATOGRAMS". after that, I run run workflow (the 3 steps are executed in few seconds) and then, I can see the chromatogram (if i adjust the scale as you can see in the GIF, because I believe, too much points are to be displayed).

we can reach on zoom when you are available so that we can compare our procedure.

Staifen, 2021-04-29:
Hi Bertrand,

I reread your message this morning and understood what you meant, my bad.
Yes it works also for me, thank you!

I can work with this in the meantime. Maybe a feature that could help for
now would be to center the plot on the integrated peak (so change "min Time
(sec)" and "max Time (sec)" to the RRT +/- 5sec, in example) whenever a
different transition is selected for plotting.

Best,
Stefano

bertrandboudaud, 2021-04-29:
Hi Staifen

Glad to see it is solved, I will close the ticket.
I will log your idea, will see. If at some point you have been confused, it's something we should learn from.

Bertrand

closed 2021-04-29 by bertrandboudaud