AutoFlowResearch/SmartPeak #378: Smartpeak not able to open sequence file on windows 10. keeps crashing!

soninikul007, 2021-06-22:
**Describe the bug**
A clear and concise description of what the bug is.

OS: Windows 10 Education version 21H1

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Smartpeak 1.7.0'
2. Click on 'File -> Load session from sequence -> map dir'
3. Scroll down to 'sequence.csv file' and select it to open
4. Smartpeak just crashes : no error

**Expected behavior**
A clear and concise description of what you expected to happen.

It should load the sequence file and allow me to perform further analysis.
sequence file attached.
[sequence.csv](https://github.com/AutoFlowResearch/SmartPeak/files/6692709/sequence.csv)

**logs**
Attach log file. Log, by default, can be found in:
 - Windows: C:\Users\user_name\AppData\Local\SmartPeak
 - Linux: /home/user_name/.SmartPeak
 - MacOS: /Users/user_name/.SmartPeak

**Screenshots**
If applicable, add screenshots to help explain your problem.

Cant take screenshot since it very quick.

**data**
Link for data files directory [e.g O:\ScientificDep\Exp42\..].
you can also attach public files but do not attach private data files to this report!

**Version information**
 - OS: [MacOS, Ubuntu, Windows]
 - Version [e.g. 1.0.0]

**Additional context**
Add any other context about the problem here.

bertrandboudaud, 2021-06-25:
Hello @soninikul007 

I don't see anything wrong with the sequence.csv that you have provided at the moment. Maybe the error comes from other files. Are you able to get some log? why do you have "no logs possible"?

In case it can help there is a way to redirect logs to another folder, by setting a user environment variable named "SMARTPEAK_LOGS". set the value to the folder that will be accessible for you. for example "c:\tmp\SmartPeak"

[image]

you can then run SmartPeak, reproduce the error and grab the log file.

[image]

Hopefully, it will help us to investigate the issue.

Regards,
Bertrand

soninikul007, 2021-06-28:
[smartpeak_log_2021-06-28_10-13-03.xlsx](https://github.com/AutoFlowResearch/SmartPeak/files/6724623/smartpeak_log_2021-06-28_10-13-03.xlsx)
please find the log file here.

bertrandboudaud, 2021-06-28:
Hello @soninikul007 

thanks for the log!

looks like an issue with your parameter file.
is it possible to provide it ?
C:\Users\soni\Documents\20210618_Robotfirstrun/parameters.csv

bertrand

soninikul007, 2021-06-28:
[parameters.xlsx](https://github.com/AutoFlowResearch/SmartPeak/files/6725872/parameters.xlsx)

i was not able to upload the csv file here somehow. so i have save it as .xlsx file. Hope this will be still okay to troubleshoot. 

Nikul

bertrandboudaud, 2021-06-28:
Hello, Nikul,

There was a tabulation instead of the comma in the 1st line of the file. 

[image]

The parsing of the parameters file usually fails 'normally', but in that situation it crashes the software. We will log this and correct later to get a proper error message.

here is attached a parameters file, with the correction, that i have zipped (you have to unzip for use in SmartPeak).
[parameters.zip](https://github.com/AutoFlowResearch/SmartPeak/files/6726806/parameters.zip)

Let us know the result.

with regards,
bertrand

soninikul007, 2021-06-29:
Thank you so much for the parameters file. it worked like a charm. Now i have another issue with smartpeak. I use the same sequence file and parameters file for running pos mode and it worked but it crashed in neg mode. see the log file attached. 
[smartpeak_log_2021-06-29_11-13-59.xlsx](https://github.com/AutoFlowResearch/SmartPeak/files/6732125/smartpeak_log_2021-06-29_11-13-59.xlsx)

bertrandboudaud, 2021-06-29:
glad to see there is some progress.

the log help just a little, it looks like something happens with AccurateMassSearchEngine.
i would suggest:

1. is your sequence file correct? is your scan_poliarity in the sequence file well set to negative?
2. check carefully the related parameters (with function AccurateMassSearchEngine).
3. change n_thread parameter to 1. It will be slower but the log will be more instructive perhaps - more sequencial - (then you can send the log back with that parameter set to 1).

please check the points above then if it has no success we can think about another way to identify the reason of this failure.

bertrand

soninikul007, 2021-06-29:
Hello Bertrand, 

yes i have tried to make sure of the point that you raised. please see the attached sequence file, parameters file and log file to see if i have missed something.

best regards,
Nikul

[smartpeak_log_2021-06-29_13-47-29.xlsx](https://github.com/AutoFlowResearch/SmartPeak/files/6733147/smartpeak_log_2021-06-29_13-47-29.xlsx)
[parameters.xlsx](https://github.com/AutoFlowResearch/SmartPeak/files/6733148/parameters.xlsx)
[sequence.xlsx](https://github.com/AutoFlowResearch/SmartPeak/files/6733149/sequence.xlsx)

bertrandboudaud, 2021-06-30:
Hello Nikul,

everything looks ok.
the most efficient way to progress would be to get access if possible to the data that you use to run the sample. is there a way to get it?

regards,
bertrand

soninikul007, 2021-06-30:
Hi Bertrand, 

zoom meeting sounds good, please email the meeting link to soni@chalmers.se

best regards,
Nikul

bertrandboudaud, 2021-07-01:
Hi Nikul,

please try SmartPeak v1.8.0 here : https://github.com/AutoFlowResearch/SmartPeak/releases/tag/v1.8.0
it should fix the issue.

with regards,
bertrand

bertrandboudaud, 2021-07-06:
The root cause was a column containing a tab instead of a semicolumn in the header, SmartPeak will be fixed to report the error, 
but it has to be corrected in the parameter file by the user anyway.

The other bug then has been verified fixed in SmartPeal 1.8.0 which is we were not handling properly a spectrum with 0 peak, which seems to happen sometimes.

closing bug, thanks :)

bertrand

closed 2021-07-06 by bertrandboudaud