AutoFlowResearch/SmartPeak #380: [Feature request] batch-export plots as images

buijt, 2021-06-25:
Hello,

One of the pain points with the SmartPeak GUI is the inability to return to the visualization plots after closing an interactive session. Currently, in order to go back and inspect the integration results after closing SmartPeak, To view the SmartPeak chromatogram plots again, I must reload the original session from the `sequence.csv` file, select and re-run the workflow, then re-open relevant tabs across the interface. My main need is to be able to quickly and easily inspect how SmartPeak decides to integrate a chromatogram from my MRM data without needing to re-process my data through the GUI.

Currently, when I notice a troublesome peak that I might want to come back to later, I will take a screenshot of the chromatogram plot in SmartPeak, but this quickly becomes unwieldy and tedious. Therefore, I would appreciate the ability if SmartPeak had the option to export specific peaks for each sample to a .png or .jpeg file for later review. I believe there would be value in being able to automatically export images of the integration + chromatogram for later inspection when using SmartPeak in headless mode (after all, headless means no GUI).

From reading the text stored in the saved .featureXML files, I suspect the integration points are being stored in them; however, I haven't found a way to view the integration curves on top of the raw chromatograms outside of SmartPeak. The .featureXML files are viewable in the OpenMS TOPPView application, but only in 2D mode (as far as I can tell). I haven't found a tutorial for being able to use TOPPView to view the integration + chromatogram, so the next best thing for me would be able to have a built-in ability to save SmartPeak plots as images directly.

Would this feature be feasible to add to SmartPeak at some point?

dmccloskey, 2021-06-28:
Hi,

Thanks for your comments and feedback!

It is possible to view the chromatogram plots after closing an interactive session.  Please try running the following workflow steps: 1) LOAD_RAW_DATA, 2) MAP_CHROMATOGRAMS, and 3) LOAD_FEATURES.  Steps 1 and 2 load in the acquired chromatographic points; Step 3 loads in the saved integrated points from the .featureXML files (assuming those have been written to disk using the workflow step STORE_FEATURES at the end of the original workflow).

Unfortunately, the graphical library we are using does not support exporting the views to .png/.jpeg/.svg etc.  This is due to technical reasons which allow for better performance on the GUI side.  

With that said, you are able to load in the raw data files as well as the .featureXML files using pyOpenMS and then plot the points using MatPlotLib.  Please see here https://github.com/dmccloskey/smartPeakPy/blob/develop/smartPeak/ui/FeaturePlotter.py for the script I was using previously that can plot and export the chromatograms to .pdf.

Please let us know how those solutions worked.  Any additional feedback is greatly appreciated.

buijt, 2021-07-01:
Thank you for your suggestions, Douglas. Both approaches work quite well. I did not explore your GitHub profile and did not know you had a Python prototype of SmartPeak among your repositories. Your `FeaturePlotter.py` code is a great guide for me to build my own plotting tools.

closed 2021-07-01 by buijt