AutoFlowResearch/SmartPeak #381: Building SmartPeak on Mac OS

soninikul007, 2021-06-29:
Hello, 

I am trying to build smart peak on my Mac OS Big Sur version 14.0, but I am getting this error.
Can anyone help me fix that.

best regards,
Nikul

[image]

bertrandboudaud, 2021-06-30:
Hello Nikul,

I think you are missing the third-party libraries that openMS uses.

in case you don't do it that way, the easiest way to recompile SmartPeak is to follow what the CI does. the script is located in
SmartPeak\.circleci\config.yml, from the ``build_macos`` section.

in your case, the contrib_build part is probably missing. but if you haven't followed that config.yml I would recommend to redo it from scratch and follow the script, instead of patching your current build work, it's only a few minutes work to redo.

with regards,
Bertrand

ahmedskhalil, 2021-09-08:
As Bertrand said, it's recommended to follow the installation instructions provided : macOS -> [STEP 2: Build OpenMS](https://github.com/AutoFlowResearch/SmartPeak#step-2-build-openms-2).

closed 2021-09-08 by ahmedskhalil