AutoFlowResearch/SmartPeak #384: Do you have nightly builds hosted somewhere?

buijt, 2021-07-01:
See title. I am interested in trying to install SmartPeak on a computing cluster, which would necessitate using the CLI from #365. I was curious if the team has nightly builds for this, or if I should compile this branch from source. I'm eager to try out the command-line mode for SmartPeak and I didn't want to wait until it is merged into the main development release.

Thanks a lot for continued help!

dmccloskey, 2021-07-02:
Good timing!  We are on a final round of internal reviews before merging it into develop and then into the next release.  I am not sure if it will make it into our release for this sprint though, but I think we can provide you with the binary once it is merged.

What operating system is the cluster running?  Linux?

We would be very interested to hear about your initial experiences using the CLA.

buijt, 2021-07-06:
Yes, the cluster is running CentOS Linux 7 (Core). I see the pull request was merged last week; fortuitous timing indeed. Thank you in advance for the Linux binaries, and I will continue to provide feedback as I run into issues.

buijt, 2021-07-07:
Hello. I see that you've released a 1.9.0 version build with the CLI; however, there is no executable for the CLI in the `bin` directory of the install path. I suspect this may be a build failure? I am sure you are already on this. I appreciate the team's work on getting this feature working!

dmccloskey, 2021-07-07:
Hi.  The current release contains the functionality for the CLI, but we still need to update our installation process to provide the executable.  This is in progress...  Hopefully, we will have it ready in a few days.

dmccloskey, 2021-07-08:
Hi Jonathan.  We made a quick release that includes the CLI executable during installation.  Please let us know how it works out!

buijt, 2021-07-08:
Thank you, I appreciate the team's quick turnaround on building the CLI binaries. I will start testing it out.

closed 2021-07-08 by buijt