AutoFlowResearch/SmartPeak #393: Parameter to modulate level/amount of smoothing applied to peaks?

buijt, 2021-07-10:
In SCIEX OS' AutoPeak peak integration algorithm, there is a pre-processing parameter option for "low" or "high" levels of peak smoothing. I am curious if SmartPeak has a similar parameter setting to modulate the level of chromatogram smoothing before applying the EMG integration model.

In the parameters shown in the GUI for the LCMS MRM Unknowns preset, I see there is a setting in `MRMFeatureFinderScoring`, `TransitionGroupPicker::peak_integration` that can take values "original" (simple integration on the raw chromatogram) or "smoothed" (which integrates into very nice Gaussian curves). However, I don't see a setting to tweak the level of smoothing applied to the peaks before the integration step.

My concern is making sure that there is limited pre-processing of the chromatogram before integration. My current assumption is that smoothing is applied before performing integration using the EMG model, unless I am mistaken regarding the flow of the SmartPeak algorithm.

Also as a side question, the original paper + methods supplement did not go into how smoothing is applied. Does SmartPeak perform a simple moving average between adjacent time points?

Thanks.

dmccloskey, 2021-07-10:
Good question. 

There are two types of smoothing: 1) Gaussian or 2) Savitzky–Golay.  The type of smoothing is handled by the parameter `PeakPickerMRM:use_gauss`; and level of smoothing for either Gaussian or Savitzky-Golay are handled by the parameters `PeakPickerMRM:sgolay_frame_length`, `PeakPickerMRM:sgolay_polynomial_order`, `PeakPickerMRM:gauss_width`, `PeakPickerMRM:peak_width`.  Additional parameters `PeakPickerMRM:signal_to_noise`, `PeakPickerMRM:sn_win_len`, `PeakPickerMRM:sn_bin_count` control the detection of the peaks.  Additional details can be found in the OpenMS documentation here https://abibuilder.informatik.uni-tuebingen.de/archive/openms/Documentation/nightly/html/classOpenMS_1_1GaussFilter.html and https://abibuilder.informatik.uni-tuebingen.de/archive/openms/Documentation/nightly/html/classOpenMS_1_1SavitzkyGolayFilter.html.

Smoothing was applied before running the EMG model in the paper, but the user can control the extent of smoothing using the parameters above.

Please let us know if you have any further questions.

buijt, 2021-07-10:
As always, thank you for your very quick response. I will look into these parameters.

closed 2021-07-10 by buijt