AutoFlowResearch/SmartPeak #405: I am trying to run the "FIAMS Unknowns" workflow using SmartPeak on a set of mzML files.

ParizadB, 2021-08-11:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**logs**
Attach all recent SmartPeak log files which can be found in the following locations :
 - Windows: C:\Users\user_name\AppData\Local\SmartPeak
 - Linux: /home/user_name/.SmartPeak
 - MacOS: /Users/user_name/.SmartPeak

**Screenshots**
If applicable, add screenshots to help explain your problem.

**data**
Link for data files directory [e.g O:\ScientificDep\Exp42\..].
you can also attach public files but do not attach private data files to this report!

**Version information**
 - OS: [MacOS, Ubuntu, Windows]
 - Version [e.g. 1.0.0]

**Additional context**
Add any other context about the problem here.

ahmedskhalil, 2021-08-12:
@ahmedskhalil

ahmedskhalil, 2021-08-12:
Hi, thanks for filing the issue. we ran the file including all the workflow steps with `sequence_rats.csv` you provided and we could not replicate the issue, have you succeeded running similar workflow steps with the same data ?

ParizadB, 2021-08-16:
Hi,
I tried with the attached workflow which extracts features from mzML files and stores them. Still got the same problem.
[feature_extraction.txt](https://github.com/AutoFlowResearch/SmartPeak/files/6992226/feature_extraction.txt)
[smartpeak_log_2021-08-16_14-11-05.csv](https://github.com/AutoFlowResearch/SmartPeak/files/6992228/smartpeak_log_2021-08-16_14-11-05.csv)

ParizadB, 2021-09-01:
Here is the system report I get after the crash.
[report.txt](https://github.com/AutoFlowResearch/SmartPeak/files/7091312/report.txt)

bertrandboudaud, 2021-09-02:
In the callstack, we can see Thread 11 has crashed

```
Thread 11 Crashed:
0   libOpenMS.dylib               	0x000000010983b773 void std::__1::__tree_remove<std::__1::__tree_node_base<void*>*>(std::__1::__tree_node_base<void*>*, std::__1::__tree_node_base<void*>*) + 35
1   libOpenMS.dylib               	0x000000010983d096 OpenMS::Logger::LogStreamBuf::addToCache_(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 966
2   libOpenMS.dylib               	0x000000010983c2b6 OpenMS::Logger::LogStreamBuf::syncLF_() + 246
3   libOpenMS.dylib               	0x000000010983d789 OpenMS::Logger::LogStreamBuf::sync() + 9
4   libc++.1.dylib                	0x00007fff203ded79 std::__1::basic_ostream<char, std::__1::char_traits<char> >::flush() + 65
5   libOpenMS.dylib               	0x0000000109a21f3e OpenMS::FeatureMap::setPrimaryMSRunPath(std::__1::vector<OpenMS::String, std::__1::allocator<OpenMS::String> > const&) + 94
6                                 	0x000000010783437b SmartPeak::PickMS1Features::process(SmartPeak::RawDataHandler&, SmartPeak::ParameterSet const&, SmartPeak::Filenames const&) const + 11995 (RawDataProcessor.cpp:2095)
```

this failure in std::__1::__tree_remove<std::__1::__tree_node_base<void*>*> seems to indicate a memory corruption. The "base call" is the PickMS1Features processor.

We can also see that some other threads (2 others) are executing also PickMS1Features at the same time. could we have concurrent access there or invalid memory access?

@ahmedskhalil does this callstack gives you some other clues? perhaps we can add more logs so that Parizad give you more details of the context.

@ParizadB it would be interesting to change the number of working threads to see if it make it pass. in the files you have given, we can see the number of threads is 29 which is quite high. we can have a try reducing it. may be we can set it to 1. It will be slow but it could be interesting to see if it still crashes.

the parameter is located in the parameters.csv file:
[image]
is it possible to change it to 1, save and re-run. if it fails, return the log and the callstack?

thank you

ParizadB, 2021-09-02:
Running the second part of the workflow with n_thread = 4 resulted in the attached stack call 
[report_thread_4.txt](https://github.com/AutoFlowResearch/SmartPeak/files/7099227/report_thread_4.txt)

closed 2021-11-16 by bertrandboudaud