AutoFlowResearch/SmartPeak #414: QC_transition_message and QC_transition_group_message do not appear in the PivotTable

buijt, 2021-08-30:
**Describe the bug**
Hello. The Report function has an ability to export a meta_value called "QC_transition_message" and "QC_transition_group_message." I understand this message to contain information regarding which QC threshold had failed the QC rules defined in featureQCComponents.csv and featureQCComponentGroups.csv. However, I notice that recently this is not included PivotTable report, but they are included in the FeatureDB report!

I believe this to be a bug where the PivotTable does not include all the Metadata as specified.

**To Reproduce**
Steps to reproduce the behavior:
1. Run a dataset that contains a featureQCComponents.csv and featureQCComponentGroups.csv, e.g. the LCMS_MRM_Unknowns example dataset on GitHub.
2. At the top of the GUI, open Actions > Report.
3. Check off all Sample Types, then in the Metadata column, check off "QC_transition_message" and "QC_transition_group_message."
4. Press "Create PivotTable.csv" and "Create FeatureDB.csv"
5. Inspect PivotTable.csv and there is no data reported for these meta_values.
6. Inspect FeatureDB.csv and scroll all the way to the right. There are data columns for "QC_transition_message" and "QC_transition_group_message."

**Expected behavior**
PivotTable contains data for "QC_transition_message" and "QC_transition_group_message."

**Version information**
 - OS: Windows 10
 - Version 1.12.0

bertrandboudaud, 2021-09-10:
Hello,

it seems we filter out all metadata with value 0.0.
@dmccloskey do you know the reason of this?

```
            if (datum.getTag() == CastValue::Type::FLOAT && datum.f_ != 0.0 && !std::isnan(datum.f_)) { // Skip NAN (replaced by 0 later)
              data_dict[sample_name].emplace(row_tuple_name, datum.f_);
              columns.insert(sample_name);
              rows.insert(row_tuple_name);
            }
```

dmccloskey, 2021-09-10:
That looks like a bug.  We want to replace NaN with a 0 but instead it just drops the row how it is written currently.

closed 2021-09-22 by dmccloskey