AutoFlowResearch/SmartPeak #415: SmartPeak crashes when given invalid directory

buijt, 2021-08-30:
**Describe the bug**
Self-explanatory title

**To Reproduce**
Steps to reproduce the behavior:
1. Open SmartPeak GUI
2. File > Open session from sequence...
3. Press "Change dir" button and provide a nonsense path name, e.g. "abc" then press "Set"
4. SmartPeak crashes

**Expected behavior**
SmartPeak gracefully surfaces an error about invalid path.

**logs**
[smartpeak_log_2021-08-30_11-06-10.log](https://github.com/AutoFlowResearch/SmartPeak/files/7078606/smartpeak_log_2021-08-30_11-06-10.log)

**Version information**
 - OS: Windows 10
 - Version 1.12.0

ahmedskhalil, 2021-08-31:
Thanks, will be fixed in the upcoming release.

ahmedskhalil, 2021-09-08:
This will be fixed in the upcoming version 1.13.0 which will be released soon.

closed 2021-09-08 by ahmedskhalil