AutoFlowResearch/SmartPeak #416: SmartPeakCLI not included in the Docker image

buijt, 2021-09-01:
Hello. I contacted the team that manages my institution's computing cluster to ask them to install SmartPeak onto our Linux server and one of the team members mentioned offhand to me that they initially attempted to build a Docker image for SmartPeak using the provided Dockerfile, but that SmartPeakCLI was unfortunately not included.

I recall a while back that the SmartPeak team needed to update the build instructions to compile SmartPeakCLI when compiling normally from source; is it possible that there was an oversight in updating the Docker build instructions?

Let me know if you need more details to look into this. I am not directly involved in the installation process for my institution's computing cluster but I can request more information if necessary.

Many thanks.

bertrandboudaud, 2021-09-02:
Dear Jonathan,

For the release, we provide an installation package (windows .exe, mac os, and deb packages for Linux). What has been done previously was to include the SmartPeakCLI in the deb package and windows, which was missing.

The dockerfile has not been updated for a while. we can plan to do it.

with regards,
bertrand

ahmedskhalil, 2021-10-21:
Please follow the Readme in : [autoflowresearch/smartpeak-cli](https://hub.docker.com/r/autoflowresearch/smartpeak-cli) .

closed 2021-11-09 by ahmedskhalil