AutoFlowResearch/SmartPeak #418: [Curiosity] How is inj_number in sequence.csv used by SmartPeak?

buijt, 2021-09-09:
Hello, I had a minor question about the inj_number column that is required to be filled out in sequence.csv

How is inj_number used in the SmartPeak application with regards to data processing? Will inj_number play a role when interspersing QC / Blank / Solvent / etc. injections between sample acquisitions? Is it used in some other data processing steps that is not triggered by the LCMS_MRM_Unknowns workflow preset (perhaps batch correction or RT correction or calibration curve generation)? 

I have experimented a little with this column and determined that the inj_number order of samples has no affect on the quantitation reported by the application (at least when my sequence consists only of Unknowns).. As a result, the inj_number column seems pointless to me right now. Since inj_number is a required column to fill out when providing sequence.csv to SmartPeak, I am curious to know the significance of the inj_number column and how it is used by SmartPeak.

### Description of my tests

I performed these tests using the same experiment with identical input files (traML, parameters, mzML), but varied the inj_number order in sequence.csv. Only Unknown samples were run, and the workflow used was LCMS_MRM_Unknowns preset.

Different sequence.csv inj_number orders:

- inj_number in the order that samples were acquired by the instrument
- inverted the inj_number so that the order goes from samples acquired last to first
- inj_number is identical across all samples (i.e. unordered)
- inj_number provided out of order from the order the samples were acquired.

I generated a PivotTable report with all meta_values from each run, sorted the table, and determined that regardless of the sample order as specified by inj_number, SmartPeak will produce identical results. So I am curious about this column.

I would appreciate an explanation about how inj_number is used. It would satisfy my curiosity. Thank you.

dmccloskey, 2021-09-10:
Thanks for your question.  The short answer is the injection number does not affect the analysis.  The injection number is to ensure that each sample is given a unique injection name for data storing and internal algorithm use.

buijt, 2021-09-10:
Thank you for the answer!

closed 2021-09-10 by buijt