AutoFlowResearch/SmartPeak #425: [Linux bug] SmartPeakCLI - Fails to load featureQCComponents.csv when featureQCComponentGroups.csv is not provided.

buijt, 2021-09-22:
**Describe the bug**
When only featureQCComponents.csv is provided to SmartPeakCLI on Linux, SmartPeak will fail to perform QC on the data. Only when featureQCComponents.csv is provided AND featureQCComponentGroups.csv is provided will SmartPeak load the QC file and check the QC rules.

This only manifests on Linux.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the LCMS_MRM_Unknowns SmartPeak example dataset
2. Rename / delete featureQCComponentsGroups.csv to something that SmartPeak will not recognize, but keep featureQCComponents.csv in the directory as-is
3. On Linux, use SmartPeakCLI to load sequence.csv and run the workflow.
4. Open resulting PivotTable.csv OR FeatureDB.csv and examine the QC_transition_pass and QC_transition_score values
5. Notice that all the values for these two columns are 1, indicating that the QC system was not invoked
6. Repeat steps on Windows, but this time there are various values in QC_transition_pass and QC_transition_score columns, indicating that QC works.

**Expected behavior**
Loads featureQCComponents.csv and computes QC_transition_pass and QC_transition_score without issue.

**logs**
Attach all recent SmartPeak log files which can be found in the following locations :
 - Windows: [Windows_smartpeak_log_2021-09-22_16-17-20.csv](https://github.com/AutoFlowResearch/SmartPeak/files/7214368/Windows_smartpeak_log_2021-09-22_16-17-20.csv)
 - Linux: [Linux_smartpeak_log_2021-09-22_15-37-07.csv](https://github.com/AutoFlowResearch/SmartPeak/files/7214343/Linux_smartpeak_log_2021-09-22_15-37-07.csv)

**FeatureDB.csv data results**
- [LCMS_MRM_Unknowns_FeatureDB_linux.csv](https://github.com/AutoFlowResearch/SmartPeak/files/7214346/LCMS_MRM_Unknowns_FeatureDB_linux.csv)
- [LCMS_MRM_Unknowns_FeatureDB_windows.csv](https://github.com/AutoFlowResearch/SmartPeak/files/7214347/LCMS_MRM_Unknowns_FeatureDB_windows.csv)

**Version information**
 - OS: CentOS Linux 7 (Core) and Windows 10
 - SmartPeak Version 1.12.0

buijt, 2021-09-24:
Small correction to my original post. The bug manifests using the CLI on both Windows and Linux. However, it does not manifest on Windows using the GUI. The attached Linux log was generated by the CLI, whereas the Windows log was generated from the GUI.

bertrandboudaud, 2021-09-28:
Hello,

We are introducing a user session file in the next release, that will allow the user to set explicitly the different files: parameters, FeatureQC files... (the old way, with one directory where lies well-named files also still work).

This bug has been corrected along with this modification.

regards,
Bertrand

closed 2021-09-28 by bertrandboudaud