AutoFlowResearch/SmartPeak #426: OpenGL issue upon compiling from source

oligegilo, 2021-09-23:
Hi, I work on a Red Hat Linux system and would like to install the latest version of SmartPeak from source but ran into issues regarding OpenGL when compiling with cmake.

cmake -DUSE_SUPERBUILD=ON -G "Unix Makefiles" -DCMAKE_BUILD_TYPE=Debug ~/SmartPeak
make -j4

works fine, but the next step

cmake -DEIGEN_USE_GPU=OFF -DUSE_SUPERBUILD=OFF -DBOOST_USE_STATIC=OFF -G "Unix Makefiles" -DCMAKE_PREFIX_PATH=$HOME/OpenMS-build/ -DPLOG_INCLUDE_DIR=$HOME/SmartPeak_superbuild/Dependencies/Source/plog/include -DIMGUI_DIR=$HOME/SmartPeak_superbuild/Dependencies/Source/imgui -DIMPLOT_DIR=$HOME/SmartPeak_superbuild/Dependencies/Source/implot -DCMAKE_BUILD_TYPE=Debug ~/SmartPeak

crashes like so:

CMake Error at /home/USER/Transcriptomics/software/cmake-3.21.0-rc2-linux-x86_64/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find OpenGL **(missing: OPENGL_opengl_LIBRARY OPENGL_glx_LIBRARY
  OPENGL_INCLUDE_DIR)**
Call Stack (most recent call first):
  /home/USER/Transcriptomics/software/cmake-3.21.0-rc2-linux-x86_64/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /home/USER/Transcriptomics/software/cmake-3.21.0-rc2-linux-x86_64/share/cmake-3.21/Modules/FindOpenGL.cmake:443 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  src/tests/class_tests/smartpeak/CMakeLists.txt:91 (find_package)

I am aware that I am missing the OpenGL header and libs - I tried to pull recommended packages in via Conda (mesa-libgl-cos6-x86_64, libglu and mesa) and link it within the DCMAKE_PREFIX_PATH but doesn't work. Could you point me into the right direction in how to properly get OpenGL to work to use SmartPeak? Thanks!

oligegilo, 2021-11-03:
Hi, last but not least we installed OpenGL globally and that did the job. I installed the Linux release yes.
However, when I finally like to build SmartPeak 'make -j4 runs' into a missing 'plog/Log.h' error - any idea how to solve this issue?

[  1%] Building CXX object _deps/googletest-build/googletest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
[  2%] Building CXX object src/smartpeak/CMakeFiles/SmartPeak.dir/source/core/ApplicationProcessor.cpp.o
[  2%] Building CXX object src/smartpeak/CMakeFiles/SmartPeak.dir/source/core/CastValue.cpp.o
[  3%] Building CXX object src/smartpeak/CMakeFiles/SmartPeak.dir/source/core/ConsoleHandler.cpp.o
In file included from /home/USER/Metabolomics/software/SmartPeak/src/smartpeak/source/core/CastValue.cpp:24:
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/core/CastValue.h:29:10: fatal error: plog/Log.h: No such file or directory
 #include <plog/Log.h>
          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/smartpeak/CMakeFiles/SmartPeak.dir/build.make:90: src/smartpeak/CMakeFiles/SmartPeak.dir/source/core/CastValue.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/USER/Metabolomics/software/SmartPeak/src/smartpeak/source/core/ConsoleHandler.cpp:23:
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/core/ConsoleHandler.h:29:10: fatal error: plog/Log.h: No such file or directory
 #include <plog/Log.h>
          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/smartpeak/CMakeFiles/SmartPeak.dir/build.make:104: src/smartpeak/CMakeFiles/SmartPeak.dir/source/core/ConsoleHandler.cpp.o] Error 1
In file included from /home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/core/RawDataHandler.h:35,
                 from /home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/core/RawDataProcessor.h:28,
                 from /home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/core/ApplicationHandler.h:27,
                 from /home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/core/ApplicationProcessor.h:26,
                 from /home/USER/Metabolomics/software/SmartPeak/src/smartpeak/source/core/ApplicationProcessor.cpp:24:
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/core/CastValue.h:29:10: fatal error: plog/Log.h: No such file or directory
 #include <plog/Log.h>
          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/smartpeak/CMakeFiles/SmartPeak.dir/build.make:76: src/smartpeak/CMakeFiles/SmartPeak.dir/source/core/ApplicationProcessor.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1097: src/smartpeak/CMakeFiles/SmartPeak.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  3%] Linking CXX static library ../../../lib/libgtestd.a
[  3%] Built target gtest
make: *** [Makefile:166: all] Error 2

bertrandboudaud, 2021-11-08:
Hi,

I never tried to compile on redhat. however, it is recommended, for ubuntu build to follow the script used by circleci:  .circleci/config.yml

on the ubuntu section, you will find there is a step named "Installing SmartPeak SuperBuild".
It is supposed to get and build some dependencies like: imgui, implot, plog, sqlite.

as you can see the command to get and build is as follow:
```
cd ~/SmartPeak
mkdir -p superbuild
cd superbuild
cmake -DUSE_SUPERBUILD=ON -G "Unix Makefiles" -DCMAKE_BUILD_TYPE=Release ~/SmartPeak
make -j4
```

you may try that (and in general follow this config.yml)

oligegilo, 2021-12-02:
Hi, thanks in fact I did the superbuild but as you already mentioned RedHat did not accept it for some reason.
For the record I solved it by locally installing imgui, implot and plog and then just pointed out the respective locations under the DCMAKE_PREFIX_PATH.

dmccloskey, 2021-12-28:
OK great to hear.  Please get back if you run into any other issues.

closed 2021-12-28 by dmccloskey