AutoFlowResearch/SmartPeak #436: What information goes into the quantitation methods file?

buijt, 2021-10-14:
With the release of the new SmartPeak user session feature, I must provide a "quantitation methods" file before I can execute the LCMS_MRM_Unknowns workflow preset. However, I do not quite understand what information should be captured in this file. 

From the example datasets, I can infer that the following columns are required:

- "IS_name" this is self-explanatory
- "component_name" also self-explanatory
- "feature_name" the example file fills this with the value "peak_apex_int" so this presumably can specify any SmartPeak meta_value, but I am not quite sure.
- "concentration_units" is self-explanatory
- "transformation_model" "transformation_model_param_slope" and "transformation_model_param_intercept" puzzle me. Where should I expect to derive this information for each component? Does Smartpeak have the capability to compute these values for me?
- The remaining columns "transformation_model_param_x_datum_min" "transformation_model_param_x_datum_max" "transformation_model_param_y_datum_min" and "transformation_model_param_y_datum_max" are filled with placeholder values, but I am not sure how these relate to / affect the transformation model.

I would appreciate if you could explain how I should fill out the quantitation methods file for my data. Thank you!

dmccloskey, 2021-10-14:
Thanks for the question.  Let me add in a few explanations below:

- "feature_name": this can be either "peak_apex_int" (peak height) or "intensity" (peak area) currently.
- "transformation_model": Currently only linear models are supported at this time.  The linear model is of the form y = mx + b where y is the concentration ratio, x is the peak height or area ratio, and m and b are the slop and intercept of the best fit line. 
 Future versions could potentially support more complicated models.
- "transformation_model_param_slope" and "...param_intercept":  These can either be calculated by SmartPeak by running `Calculate_Quantitation_Methods` based on a given set of calibration standards provided in the file `StandardsConcentrations.csv`.  Alternatively, the user can provide these values if they are already known (e.g., at the extreme end if a single point calibration based on a known concentration of internal standard is used).
- "..._min" and "...max": These set the minimum and maximum x and y values to consider when fitting the linear model

Hopefully, some useful references:
- `QuantitationMethod` class documentation https://abibuilder.informatik.uni-tuebingen.de/archive/openms/Documentation/nightly/html/classOpenMS_1_1AbsoluteQuantitation.html#details
- Example for fitting a calibration curve based on user supplied calibration standards https://github.com/AutoFlowResearch/SmartPeak/tree/develop/src/examples/data/LCMS_MRM_Standards

Please let me know if that is helpful (or not).

buijt, 2021-10-15:
Thank you for the quick response. I will look into this.

closed 2021-12-28 by dmccloskey