AutoFlowResearch/SmartPeak #453: SmartPeak GUI Crashing on MacOS

gkreder, 2021-11-09:
**Describe the bug**
I'm running the SmartPeak GUI on an Intel Macbook Pro 16" MacOS Big Sur 11.6.1. When I try to run a workflow on the sample data, the program immediately closes (crashes?). No output files are created. This happens as soon as a I click "Run Workflow" on the popup window asking for the input mzML, input features, and output features folders 

**To Reproduce**
Steps to reproduce the behavior:
1. Open SmartPeak GUI
2. File -> New Session
3. Select session directory from Sample Data and click "Ok" (have tried "LCMS_MRM_QCs", "LCMS_MRM_Standards", "LCMS_MRM_Unknowns", and "HPLC_UV_Standards"). 
4. Actions -> Run Workflow
5. Confirm mzML folder, Input features folder, and Output features folder click "Run workflow"
6. Program crashes (happening every time)

**Expected behavior**
I expect the workflow to run at this point and to create output files/results

**logs**
SmartPeak crash log
[smartpeak_log_2021-11-08_15-27-04.log](https://github.com/AutoFlowResearch/SmartPeak/files/7503435/smartpeak_log_2021-11-08_15-27-04.log)

Apple Crash Report Log
[smartpeak_apple_crash_log_211109.log](https://github.com/AutoFlowResearch/SmartPeak/files/7503437/smartpeak_apple_crash_log_211109.log)

**data**
I am using the SmartPeak sample data for now

**Version information**
 - OS: MacOS 11.6.1 (Macbook Pro 16" Intel i7)
- SmartPeak - 1.16.0

bertrandboudaud, 2021-11-10:
Looks like to be the same as  (https://github.com/AutoFlowResearch/SmartPeak/issues/405.)

We are working on MacOs as well but cannot reproduce it with our setup. It seems that a specific os version or hardware is necessary to reproduce it.

However it seems that we can reproduce it using the CI, so we can work on that.

gkreder, 2022-01-11:
Hello, 

I've downloaded SmartPeak release 1.18.0, and I seem to be having the same issue still. The GUI crashes as described above. I tried running on MacOS Big Sur 11.6.1 (Intel Macbook Pro 16") then upgraded to MacOS Monterey 12.1. The issue is happening in both MacOS versions. I have attached the log of the latest crashed run here

[smartpeak_log_2022-01-11_09-51-29.log](https://github.com/AutoFlowResearch/SmartPeak/files/7845131/smartpeak_log_2022-01-11_09-51-29.log)

Is there anything I might be doing wrong here?

Thank you

bertrandboudaud, 2022-01-11:
Hello,

Indeed, it seems to be not fixed. I re-open it.
We will investigate.

Bertrand

bertrandboudaud, 2022-01-26:
While I am able to reproduce it with SmartPeak 1.16, I cannot get that crash with 1.18.
@gkreder do you mind sending me some call stacks, (crash logs)1 or more if you have, if it's still on your hard drive ?

That would help a lot.

I'll continue to try to find some conditions to make it crash.

Thanks
Bertrand

gkreder, 2022-02-09:
Hi @bertrandboudaud thanks for checking. Other than the SmartPeak log I posted above, are there any other logs that would be especially useful? Where could I find them? I can't seem to find any Mac system logs or crash reports but I may be looking in the wrong place...

bertrandboudaud, 2022-02-14:
Hi @gkreder it would be nice to have the call stacks reports.

These are in this directory:
/Users/your_user_name/Library/Logs/DiagnosticReports

and have .is name

[image]

in the finder, Library is a hidden name so to show it you have to Press Command + Shift + . (period) 

but they are deleted regularly, so you may have to reproduce the crash 2 or 3 times again to get these files (it would be nice to have 2-3 call stacks files).

Actually, another thing that I had noticed is that if you install SmartPeak by drag and dropping the app in the Applications directory but forgetting to also drag the **share** folder, that comes with the package you may have a crash when running the workflow, at the 1st second. It is actually not a crash but just a wild exit that does not produce any call stack file so in that case you'll not find anything in the directory above.

So may be you can confirm that buy drag and dropping the share folder in the app. if this is the problem then we will try to arrange that to embed the share folder more properly.

[image]

thank you
bertrand.