AutoFlowResearch/SmartPeak #462: Scripting issues during make

oligegilo, 2021-12-02:
Hi all, after I managed to get cmake working, make ran into some scripting issues - I guess it's because I'm working on a RedHat system which sometimes needs a special treatment. The output of make is like below. Would be great if that could be adjusted so that make succeeds. Thank you!

[  1%] Building CXX object _deps/googletest-build/googletest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
[  1%] Building CXX object src/smartpeak/CMakeFiles/SmartPeak.dir/source/core/CastValue.cpp.o
[  2%] Building CXX object src/smartpeak/CMakeFiles/SmartPeak.dir/source/core/ApplicationProcessor.cpp.o
[  3%] Building CXX object src/smartpeak/CMakeFiles/SmartPeak.dir/source/core/ConsoleHandler.cpp.o
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/source/core/CastValue.cpp: In member function ‘bool SmartPeak::CastValue::is_less_than(const SmartPeak::CastValue&, bool) const’:
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/source/core/CastValue.cpp:217:16: error: ‘transform’ is not a member of ‘std’
           std::transform(s_.begin(), s_.end(), a_lowercase.begin(), ::tolower);
                ^~~~~~~~~
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/source/core/CastValue.cpp:218:16: error: ‘transform’ is not a member of ‘std’
           std::transform(other.s_.begin(), other.s_.end(), b_lowercase.begin(), ::tolower);
                ^~~~~~~~~
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/source/core/CastValue.cpp: In member function ‘bool SmartPeak::CastValue::is_greater_than(const SmartPeak::CastValue&, bool) const’:
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/source/core/CastValue.cpp:252:14: error: ‘transform’ is not a member of ‘std’
         std::transform(s_.begin(), s_.end(), a_lowercase.begin(), ::tolower);
              ^~~~~~~~~
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/source/core/CastValue.cpp:253:14: error: ‘transform’ is not a member of ‘std’
         std::transform(other.s_.begin(), other.s_.end(), b_lowercase.begin(), ::tolower);
              ^~~~~~~~~
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/source/core/CastValue.cpp: In member function ‘bool SmartPeak::CastValue::is_equal_to(const SmartPeak::CastValue&, bool) const’:
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/source/core/CastValue.cpp:287:14: error: ‘transform’ is not a member of ‘std’
         std::transform(s_.begin(), s_.end(), a_lowercase.begin(), ::tolower);
              ^~~~~~~~~
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/source/core/CastValue.cpp:288:14: error: ‘transform’ is not a member of ‘std’
         std::transform(other.s_.begin(), other.s_.end(), b_lowercase.begin(), ::tolower);
              ^~~~~~~~~
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/source/core/CastValue.cpp: In lambda function:
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/source/core/CastValue.cpp:314:16: error: ‘transform’ is not a member of ‘std’
           std::transform(l.begin(), l.end(), a_lowercase.begin(), ::tolower);
                ^~~~~~~~~
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/source/core/CastValue.cpp:315:16: error: ‘transform’ is not a member of ‘std’
           std::transform(r.begin(), r.end(), b_lowercase.begin(), ::tolower);
                ^~~~~~~~~
make[2]: *** [src/smartpeak/CMakeFiles/SmartPeak.dir/build.make:90: src/smartpeak/CMakeFiles/SmartPeak.dir/source/core/CastValue.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/USER/Metabolomics/software/SmartPeak/src/smartpeak/source/core/ConsoleHandler.cpp:23:
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/core/ConsoleHandler.h:176:27: error: ‘RollingFileAppender’ is not a member of ‘plog’
     std::shared_ptr<plog::RollingFileAppender<plog::CsvFormatter>> m_file_appender;
                           ^~~~~~~~~~~~~~~~~~~
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/core/ConsoleHandler.h:176:27: note: suggested alternative: ‘ConsoleAppender’
     std::shared_ptr<plog::RollingFileAppender<plog::CsvFormatter>> m_file_appender;
                           ^~~~~~~~~~~~~~~~~~~
                           ConsoleAppender
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/core/ConsoleHandler.h:176:53: error: ‘CsvFormatter’ is not a member of ‘plog’
     std::shared_ptr<plog::RollingFileAppender<plog::CsvFormatter>> m_file_appender;
                                                     ^~~~~~~~~~~~
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/core/ConsoleHandler.h:176:53: error: template argument 1 is invalid
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/core/ConsoleHandler.h:176:65: error: expected unqualified-id before ‘>’ token
     std::shared_ptr<plog::RollingFileAppender<plog::CsvFormatter>> m_file_appender;
                                                                 ^~
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/core/ConsoleHandler.h:177:49: error: ‘TxtFormatter’ is not a member of ‘plog’
     std::shared_ptr<plog::ConsoleAppender<plog::TxtFormatter>> m_console_appender;
                                                 ^~~~~~~~~~~~
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/core/ConsoleHandler.h:177:49: error: template argument 1 is invalid
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/core/ConsoleHandler.h:177:61: error: template argument 1 is invalid
     std::shared_ptr<plog::ConsoleAppender<plog::TxtFormatter>> m_console_appender;
                                                             ^~
In file included from /home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/core/ApplicationHandler.h:26,
                 from /home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/core/ApplicationProcessor.h:26,
                 from /home/USER/Metabolomics/software/SmartPeak/src/smartpeak/source/core/ApplicationProcessor.cpp:24:
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/core/Filenames.h:89:18: warning: class ‘SmartPeak::Filenames’ is implicitly friends with itself
     friend class Filenames;
                  ^~~~~~~~~
In file included from /home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/core/RawDataProcessor.h:27,
                 from /home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/core/ApplicationHandler.h:27,
                 from /home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/core/ApplicationProcessor.h:26,
                 from /home/USER/Metabolomics/software/SmartPeak/src/smartpeak/source/core/ApplicationProcessor.cpp:24:
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/core/MetaDataHandler.h:125:69: warning: missing initializer for member ‘tm::tm_gmtoff’ [-Wmissing-field-initializers]
     std::tm acquisition_date_and_time = { 0, 0, 0, 1, 0, 0, 0, 0, 0 }; // Need to start at Day 1 of the month
[... 180 lines cut ...]
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/io/csv.h:152:50: warning: conversion from ‘size_t’ {aka ‘long unsigned int’} to ‘int’ may change value [-Wconversion]
                                 return std::fread(buffer, 1, size, file);
                                        ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/io/csv.h: In member function ‘virtual int io::detail::NonOwningIStreamByteSource::read(char*, int)’:
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/io/csv.h:169:49: warning: conversion from ‘std::streamsize’ {aka ‘long int’} to ‘int’ may change value [-Wconversion]
                                 return in.gcount();
                                        ~~~~~~~~~^~
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/io/csv.h: In member function ‘virtual int io::detail::NonOwningStringByteSource::read(char*, int)’:
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/io/csv.h:185:62: warning: conversion from ‘long long int’ to ‘int’ may change value [-Wconversion]
                                         to_copy_byte_count = remaining_byte_count;
                                                              ^~~~~~~~~~~~~~~~~~~~
[... cut ...]
/home/USER/Metabolomics/software/SmartPeak/src/smartpeak/include/SmartPeak/io/csv.h:1099:22: warning: unnecessary parentheses in declaration of ‘row’ [-Wparentheses]
                 char*(row[column_count]);
                      ^
[  6%] Linking CXX static library ../../../lib/libgmockd.a
[  6%] Built target gmock
[  7%] Building CXX object _deps/googletest-build/googlemock/CMakeFiles/gmock_main.dir/src/gmock_main.cc.o
[  7%] Linking CXX static library ../../../lib/libgmock_maind.a
[  7%] Built target gmock_main
make[1]: *** [CMakeFiles/Makefile2:1097: src/smartpeak/CMakeFiles/SmartPeak.dir/all] Error 2
make: *** [Makefile:166: all] Error 2

oligegilo, 2021-12-03:
Alright, there you go:
gcc (GCC) 8.3.1 20191121 (Red Hat 8.3.1-5)

dmccloskey, 2021-12-03:
Thanks for the information.  I am thinking that it is not using the C++17 library.  Can you try compiling using the flag `-std=c++17`?