AutoFlowResearch/SmartPeak #490: With latest version of OpenMS, pickMS2Features test fails

bertrandboudaud, 2022-01-13:
**Describe the bug**
unit test pickMS2Features fails. seems to be a consequence of an update from OpenMS.

**logs**
[pickMS2Features_failed.txt](https://github.com/AutoFlowResearch/SmartPeak/files/7861555/pickMS2Features_failed.txt)

**Version information**
1.18, all platforms

bertrandboudaud, 2022-01-13:
Seems to be introduced with this commit:

https://github.com/OpenMS/OpenMS/commit/738747e5279dd2600c565f6e7bb71f6e827aa585

dmccloskey, 2022-01-13:
I took a look at the commit and I don't see anything that would affect the OpenMS methods called in the pickMS2Features method.

bertrandboudaud, 2022-01-14:
Sorry , It comes from this commit.
https://github.com/OpenMS/OpenMS/pull/5734

dmccloskey, 2022-01-15:
The changes proposed in the commit are quite beneficial and solve a long standing issue with the peak picker.  I would imagine it would cause some changes in the number of overall picked features as measured by our unit tests.

Do any of the other tests fail besides the "RawDataProcessor.pickMS2Features" test?  Do the examples also pass?

bertrandboudaud, 2022-01-17:
It seems not so i will update with the latest values.

dmccloskey, 2022-01-17:
Sounds good to me.

closed 2022-01-17 by dmccloskey