AutoFlowResearch/SmartPeak #506: Use ImPlot zooming feature

bertrandboudaud, 2022-01-21:
We want to use the facilities provided by ImPlot to navigate into the chromatograms and the spectrograms.

dmccloskey, 2022-01-21:
The only complications I can think of would be keeping the sliders that determine the ranges of x axis values in sync. Perhaps even allowing the user to reset to the original view with some sort of "reset" layout button would also be a good combo?

bertrandboudaud, 2022-01-21:
Yes, actually we can remove the sliders at all. with the Calibration curve improvement that is WIP, we can see that using the mouse wheel to zoom out/in, and have the box selection is very convenient.
And as for the Calibration curve (the WIP), and as you suggest, we can have a reset button indeed to reset the view.

dmccloskey, 2022-01-21:
That sounds good on my end 👍

closed 2022-01-26 by dmccloskey