AutoFlowResearch/SmartPeak #559: Update the Docker images?

buijt, 2023-10-09:
Hello AutoFlowResearch team,

I hope that you are all doing well. I have been using one of the earliest versions of SmartPeakCLI deployed on my HPC cluster for a couple years now, and I thinking to update it to the latest version of SmartPeakCLI to see what improvements have been made to the software since then. 

I pulled the latest version of SmartPeakCLI from [DockerHub](https://hub.docker.com/u/autoflowresearch) and I determined that the version of SmartPeakCLI contained within is 1.15.0, not the current version 1.27.0, and I wonder if I could trouble you to update the Dockerfile for SmartPeak.

Thank you!

dmccloskey, 2023-11-01:
Hi. Thanks for your usage and interest in the SmartPeak project. Yes, we can update the image.

buijt, 2023-11-01:
Thanks a lot